u-root/u-bmc #4: Add RTC to Quanta Leopard platform

bluecmd, 2018-09-15:
There is an RTC somewhere on the server.

1. Find what it is, and on which bus it is on
2. Enable it in the device tree

bluecmd, 2018-09-23:
Thinking about this some more. Maybe this is just all done by the PCH?

I see that facebook/openbmc does sync from the ME, which I don't really like. Ideally there would be some source we can query when everything is shut off.

bluecmd, 2018-09-23:
@shenki How does OpenBMC handle RTC clock? I tried looking at the code and see some past discussions but it didn't jump at me how it's done.

shenki, 2018-09-24:
In terms of providing it to the host? I don't know much about openbmc's handling of this.

On OpenPower host firmware queries the current time from the BMC at host boot time. This happens over IPMI BT. The source of this is the current time according to Linux. 

Linux on the BMC grabs the current time/date from the RTC over i2c at BMC boot. All of the systems I've seen have a i2c attached RTC that has battery backup. ASPEED does have a RTC in SoC, but it loses state when the BMC loses power, so it's not very useful.

bluecmd, 2018-09-24:
Thanks! I was curious about how the BMC gets its time - but you answered that.
 
I'll dig through some board schematics for other boards and see if I can figure it out.

Fwiw, on Yosemite I recall the RTC battery to be on each of the 4 host blades and I don't think the aspeed part had any.

bluecmd, 2018-09-24:
This is what the Tioga Pass requires:

## 8.17 BMC Time Sync

### During BMC initialization:
* If it’s the 1st time being powered on via an AC source (G3 exit), the BMC should receive
its initial RTC time from PCH RTC. Otherwise, the BMC should keep its own RTC.

### During BMC running time:
* If the system is in S0 and the BIOS POST has completed, the BMC should sync with
PCH RTC every hour
* If the system is in S5, the BMC should retrieve the NTP UTC following the OEM
command setting of NTP periodical sync interval (default =2x per hour)
* Log NTP status time sync failed event if a mismatch of > ±5 seconds is found, or if BMC
cannot obtain NTP time after 3x retries. Retry interval is set to 30 seconds by default. 

I have to research what the PCH can do when exiting G3 but my understanding is that it is very limited. I'm considering being very NTP centric instead, given that you need network connectivity to query the API anyway, having NTP should be fine. Reading the RTC from the PCH as a fallback, and potentially even power up the machine to do so should also work, but that's heading down a path of surprising behavior.

bluecmd, 2018-09-25:
Pending discussion in https://github.com/u-root/u-root/issues/961

bluecmd, 2018-09-26:
I've requested a vendor pool for `x.u-root.pool.ntp.org` as is required to ship with NTP pool defaults in apps.

closed 2018-09-29 by bluecmd

bluecmd, 2018-10-15:
RTC is read up on boot but not set by us, will file an issue for that.

bluecmd, 2018-10-15:
Pending https://github.com/u-root/u-root/issues/982

PR #69 (2018-09-29): Set RTC using roughtime and NTP
See comment in config/config.go how this is meant to work, but
essentially use roughtime as an authenticated time source and NTP to
tune the results of roughtime.

Fixes #4.
commit d1807722 (2018-09-29): Set RTC using roughtime and NTP

See comment in config/config.go how this is meant to work, but
essentially use roughtime as an authenticated time source and NTP to
tune the results of roughtime.

Fixes #4.