u-root/u-bmc #9: Clean checkout fails due to empty u-root directory

osresearch, 2018-09-15:
There is no `u-root/Makefile`, so a clean checkout fails on the first `make -C u-root` step.

osresearch, 2018-09-15:
Instructions should mention the need to do a `git submodule init && git submodule update`.

The path to the `openbmc/linux` tree also assumes an ssh key at github; the https repo should be used instead.

rjoleary, 2018-09-15:
The Linux kernel takes forever to build, so we also need to check it out with `--depth=1`.

rjoleary, 2018-09-15:
Git submodules is being difficult. I think we should have a `make depedencies` command.

rjoleary, 2018-09-16:
If we pin the kernel release to a tag rather than a commit hash, we could use something like this in the makefile:

```
git clone --depth=1 --branch=v4.17 https://github.com/openbmc/linux.git
```

bluecmd, 2019-04-03:
Linux does take a long time to check out, but you can also download a snapshot if you want like what circleci does - so I don't think this is a huge issue right now.

closed 2019-04-03 by bluecmd