u-root/u-bmc #26: Wishlist: Graphical interface for gpiowatcher

bluecmd, 2018-09-16:
I've been thinking that having a way to stream events over console and render it graphically (like how logic analysers do) would be really nice when porting new boards.

rjoleary, 2018-09-16:
I wonder if you can make a server which parses /dev/ttyUSB0 (or whichever device) and coverts to one of the formats accepted by [prometheus](https://prometheus.io/).

bluecmd, 2018-09-16:
While prometheus is a sampled system made for way higher sample rates, that did give me an idea. We could be using InfluxDB (which is event based rather than sampled) to allow for either local capture and processing, and also offer a super-simple cloud version where you just run gpiowatcher and it streams all the state changes to InfluxDB. Interesting.

bluecmd, 2019-04-08:
Unlikely to happen any time soon, re-open if needed down the road.

closed 2019-04-08 by bluecmd