u-root/u-bmc #27: Hack-mode

bluecmd, 2018-09-16:
Right now we compile everything and require a reflash every time something changes.

Given that u-root has a compilation mode for on the fly builds, investigate if it's possible to use that to be able to modify the source on the device to shorten the development iterations.

rjoleary, 2018-09-16:
If you're looking at using hacking with Go from the shell, you can also look into:

- u-root's run command: https://github.com/u-root/u-root/blob/master/xcmds/run/run.go#L3
- gomacro REPL: https://github.com/cosmos72/gomacro

bluecmd, 2018-10-02:
I'll close this as infeasible. It goes against the signed boot stuff as well and would diverge a lot, it was a cool idea but will probably not work out.

closed 2018-10-02 by bluecmd