u-root/u-bmc #32: Submit Leopard kernel patches upstream

shenki, 2018-09-17:
They ASPEED kernel maintainer eagerly awaits your device tree patch. Ping me on irc if you need help making a kernel submission.

If you have other patches you're using then we should either get them upstream, or in the openbmc tree if they are not yet ready.

bluecmd, 2018-09-17:
Cool! I have no idea if it works fully yet and there are about 10'000 (I approximate) I2C devices I would like to add as well, but I'd love to keep this open and take you up on the offer later.

Unless you want to take it in iterations with a disclaimer that it's not complete?

shenki, 2018-09-17:
You wouldn't be the first to submit an incomplete device tree. I encourage submitting what you have now, and we can add in the rest of the devices as we get them working.

bluecmd, 2018-09-17:
Sounds good, I'll clean it up a bit and submit it to OpenBMC's Linux fork.

bluecmd, 2018-10-19:
@shenki Now when I found the temperature sensors, I think the https://github.com/u-root/u-bmc/blob/master/platform/quanta-f06-leopard-ddr3.dts is good to upstream.

How should I proceed?