u-root/u-bmc #76: Upgrade U-boot

bluecmd, 2018-10-03:
We're using OpenBMC's u-boot which is based on 2016. This means that there is no chance that we can upstream any u-boot modifications we would like to do.

u-boot mainline does not support AST24xx series so that would have to be ported, but the good thing is that it's not that much stuff we require from the CPU - so it could maybe be not that hard.

Still, collecting all the contributor material and making sure it's all licensed properly etc is going to be a time sink.

shenki, 2018-10-04:
@legoater has recently re-started our efforts to upstream support for the aspeed socs. We've got flash and network drivers on the u-boot mailing list at the moment, and have some in-progress patches for NCSI.

We're focused on the ast2500 initially.

bluecmd, 2018-10-04:
@shenki Excellent, thanks for the update. I might be able to join that effort somehow if that becomes the most juice fruit to pick from the TODO basket.

shenki, 2018-10-04:
https://patchwork.ozlabs.org/project/uboot/list/?series=&submitter=68548&state=&q=&archive=&delegate=

legoater, 2018-10-05:
The difficult part of the ast2400 u-boot support is the dram training. One thing we could do would be to isolate the orignal code in its assembly file and use it as it is. The other drivers should not be too hard 
as the ast2400 and ast2500 SoC are quite similar.  Well, there  will be some work but we already have done it on Linux and QEMU.

closed 2019-02-04 by bluecmd