u-root/u-bmc #84: Consider splitting NCSI netlink functionality into own package

mdlayher, 2018-10-04:
Hello again!

I had previously suggested that we split the NCSI netlink interaction code into its own package to enable further reuse in the Go ecosystem. I'd be happy to do the work and also write some tests (with genltest package) to lock in its behavior.

For now this could live under "u-bmc/pkg/nscinl" or similar. I suggest the "nl" suffix because, if I understand, NCSI functionality can be manipulated with specially crafted Ethernet frames as well. I don't know how this all ties together.

Let me know if you'd like me to work on this! Thanks again.

bluecmd, 2018-10-04:
I agree 100% :-).

Please do what you think is right. I'll be happy to review whatever you propose.

bluecmd, 2018-10-04:
Oh! There is some activity on merging some OEM stuff regarding NC-SI. @shenki knows more. I'm kind of waiting on that for implementing more NC-SI stuff like reading MAC addresses, and reading sensors from the nic.

mdlayher, 2018-10-04:
Awesome. Looking forward to checking it out!  I'll probably have a PR ready for review tomorrow.

bluecmd, 2018-11-03:
@mdlayher There is absolutely no rush on this, but for transparency I'm removing this from the v1.0 project to prune the milestone a bit. NC-SI in userspace is right now monitoring only so even if we do a v1.0 before this completes there should be no problem changing the code structure to whatever you have in mind.

mdlayher, 2018-11-04:
Sounds good, apologies for the radio silence. Going through a job transition and taking some time off. Hope to pick this back up in a couple of weeks.