u-root/u-bmc #97: Reproducible builds

bluecmd, 2018-10-25:
Long term we want reproducible builds.

I spent today looking at OpenEmbedded and while the developers claim it's not very opinionated, it kind of is. As far as I see it really needs to be **the** build system in use or things will just not work. They do not seem to support in-tree sources either, which will be a mess if we have to split everything up.

Not seeing any easy way forward, I'll keep the status quo of recommending people to install whatever ARM compiler comes with their OS and use that one. In the future LinuxBoot etc will need to figure this out as well, so there might be opportunities to mimic there.

rminnich, 2018-10-25:
seems to me we could adopt the coreboot crossgcc stuff. It's been in use for years and works very well.

bluecmd, 2018-10-25:
Makes sense to me. It seems to have Kconfig as well which is something I wanted to try using u-bmc so that might be a good solution.

osresearch, 2019-10-02:
The boot and full kernels have build times embedded in the version strings, making them non-reproducible.