u-root/u-bmc #115: Use Aspeed UART routing control driver

bluecmd, 2018-11-12:
Right now we implement this as memory writes like this:

https://github.com/u-root/u-bmc/blob/f6c60daa5817a10f5a7a3fec3f9203546629b09a/platform/quanta-f06-leopard-ddr3/pkg/platform/platform.go#L112-L116

Using [this patch](https://patchwork.ozlabs.org/patch/969238/) it would be instead writing to sysfs.

Keno, 2019-06-05:
Activating the clocks for the pass through uarts (the next couple of lines) is probably related. Ideally we'd declare those in the .dts, so the kernel driver takes care of setting those all up, without enabling access from user space (since in the pass-through case those registers are poked directly by the host over LPC).

osresearch, 2019-10-04:
I've modified the `platform/ast2400/boot/platform_g4.S` file to enable the clocks and the routes so that the x86 serial port is available on the db9 immediately and also re-routed the BMC console to the COM2. https://github.com/osresearch/u-bmc/commit/8b7d1ae02a33bbfda7487b6a2162a3317a355f00#diff-f2bc3e1c26b9da867e3030c3a41fd7a9R359

c0d3z3r0, 2019-10-28:
This should be somehow possible via pinctrl dts entries for the bootloader kernel, too