u-root/u-bmc #120: User-land integration with OpenBMC

bluecmd, 2018-12-13:
It probably wouldn't be that hard to provide a OpenBMC daemon based on the platform specific stuff for u-bmc. That way one could for any supported u-bmc platform choose to build either full u-bmc or OpenBMC w/ u-bmc userland.

That way u-bmc would directly help with adding boards to OpenBMC, and companies that are using OpenBMC wouldn't have to migrate everything over. Full u-bmc would still be able to innovate on the things that matters for it while keeping OpenBMC compatibility.

GanShun, 2018-12-13:
I don't really know anything about OpenBMC, but are there baked in systemd assumptions with openbmc? How would you work around that? I seem to remember systemd insists on being pid 1 too, but I could be wrong.

bluecmd, 2018-12-13:
@GanShun I elaborated above since I got the same question on slack :-). The idea is to replace whatever implements for example this protocol: https://github.com/openbmc/phosphor-dbus-interfaces/tree/master/xyz/openbmc_project/Chassis

GanShun, 2018-12-13:
I see