u-root/u-bmc #121: Static network configuration support

bluecmd, 2018-12-15:
To aid in development, like VLAN, support having a file like /etc/network which disables all DHCP/RA automagic and replaces it will static IPv4/IPv6 and FQDN

rminnich, 2018-12-16:
one of the early goals of u-root, never clearly stated, was to try our best to avoid having a /etc at all; we are trying to break with past habits. We've yielded in cases where there was almost no choice (nichrome needs some things we don't have yet, and those things need /etc/passwd), but let's think on this a little. Is there any way to do this without /etc? Should we be using, e.g., u-boot environment variables? Are we sure we need /etc/network? If we do, can it be JSON? Also, if you create /etc/network, you WILL collide with other tools -- can we call it something other than /etc?

bluecmd, 2018-12-17:
Sure, we can call it /config. I was thinking of having the format be text proto or YAML. The good thing about text proto is that it has a schema so it will always be easy to read how the document should be formatted, but maybe it's too arcane.