u-root/u-bmc #133: Add AST2050 platform (KGPE-D16/KCMA-D16 motherboards)

tlaurion, 2019-02-07:
This OpenBMC [port](https://www.raptorengineering.com/coreboot/kgpe-d16-bmc-port-status.php) for the KGPE-D16 and KCMA-D8 works but is left unmaintained.

PCB plans for the ASMv4 have [been open sourced](https://github.com/DanielG/asmb4/). 
PCB can be ordered online [here](https://oshpark.com/shared_projects/jZLoDQ3Y).

Let's remember that the KGPE-d16 (and KCMA-d8) are the last FSF' RYF certifiable x86 motherboards, containing zero binary blobs.

The KGPE-Dq6 has[ been ported to Heads](https://github.com/osresearch/heads/issues/134).

bluecmd, 2019-02-07:
Thanks for the issue, definitely seems like a cool board to support.

From the old port there are some data available: [1](https://git.raptorengineering.com/git/ast2050-yocto-openbmc/tree/meta-raptor/meta-asus/recipes-utils/openbmc-gpio/files/data/asus-BMC-GPIO.csv) [2](https://git.raptorengineering.com/git/ast2050-yocto-openbmc/tree/meta-raptor/meta-asus/recipes-utils/lm_sensors/files/asus.conf) [3](https://git.raptorengineering.com/git/ast2050-yocto-openbmc/tree/meta-raptor/meta-asus/recipes-core/fan-ctrl/fan-ctrl/set_fan_speed.sh) 

[This email](https://lists.denx.de/pipermail/u-boot/2009-January/045481.html) says the AST2050 uses ARM926EJ which should be what the AST2400 series uses. However, I'm not sure if it's so close that you could boot a Linux kernel using the AST2400 configuration, and I'm not sure what the AST2050 mainline situation is. The linked kernel appears to be 2.6.x series, which sadly makes it quite a burden to update.

I would say that the next step is to look at how the QEMU and Kernel situation is in terms of this particular SoC and then make sure those are supported upstream.

tlaurion, 2019-02-16:
@bluecmd : what would be needed to port to this board from knowledgeable people? HW?

bluecmd, 2019-02-21:
HW ideally, and emulator like qemu would also be acceptable.

tlaurion, 2019-03-24:
@bluecmd 

> I would say that the next step is to look at how the QEMU and Kernel situation is in terms of this particular SoC and then make sure those are supported upstream.

Please point me to a minimal starting point and i'll try to test this myself!

bluecmd, 2019-03-24:
Sure, I'd love to help. Are you thinking of pursuing hardware or emulation?

In either case, compiling and getting a modern Linux kernel to boot on the BMC SoC is a good first task. If you managed to do that everything else is just hooking it up to the build system.

One good first step might be to build an u-boot image and burn it to the BMC and make sure you have access to a bootloader for debugging. u-bmc is not using u-boot by itself, but getting u-boot up is a standard thing for these types of SoCs so that should be a good start.

bluecmd, 2019-04-08:
Coreboot seems to have ast2050 support as well, so might be a good lead: https://github.com/coreboot/coreboot/tree/master/src/drivers/aspeed/ast2050

tlaurion, 2019-04-09:
@bluecmd hardware for sure!

_Sent from my Galaxy S3 using [FastHub-Libre](https://github.com/thermatk/FastHub-Libre/)_

pkubaj, 2019-10-03:
The drivers for ast2050 are out-of-tree and only for 2.6.29.6. I tried to upgrade the kernel and managed to get only 2.6.30.10 (https://github.com/pkubaj/ast2050-linux-kernel/tree/linux-2.6.30.10). 2.6.31 compiles, but doesn't boot and I have no JTAG soldered, so can't check why it doesn't boot.

Newer kernels differ a lot, the driver has been completely rewritten and so would have to support for ast2050. AFAIR the oldest supported ast is ast2400.