u-root/u-bmc #135: Create a development platform on Raspberry Pi

bluecmd, 2019-03-10:
Raspberry Pi could serve as a nice development platform.

It could also be useful for some things, like:

- Attach a PWM fan and a sensor and let it control the cooling of whatever you want
- Remote button pusher for some random appliance
- UART console server

Stretch goal: Support open platforms like Banana Pi as well. However, given the community around RPi I think RPi is the priority target.

bluecmd, 2019-03-12:
Most likely the hardest part of this is to figure out what the SPI flash story is. RPi doesn't use SPI flash by default (but does support it) and we don't want the development taget to differ too much from the real deal.

Would selling u-bmc SPI flash-boards as an add-on be the best way forward? Maybe, seems like it would remove some of the appeal to quickly get it up and running. Maybe another platform is better?

rjoleary, 2019-03-28:
Can the SPI flash be connected to the RPi with a breadboard and a few wires? If so, I think it would suffice to have a tutorial, wiring diagram and a short list of components to purchase. There are already RPi kits which come with most of the materials (maybe not the SPI flash).

I've run u-root on RPi before (running off the SD card). How were you thinking of booting u-bmc on the RPi?

bluecmd, 2019-03-29:
Yep, it would be like 6 wires or something like that. You're probably right that a BOM should be enough.

I was thinking of having a shim on an SD card that just jumps to the bootloader on the SPI flash.