u-root/u-bmc #186: Support Supermicro X11-SSH board

bluecmd, 2019-09-05:
@Keno has made excellent progress in porting u-bmc to this board.

This is the tracking bug for getting it merged.

osresearch, 2019-09-05:
I've forward ported his changes to u-root:master in this tree https://github.com/osresearch/u-bmc/tree/kf/x11

There are a few weirdnesses that we can document:

* The x86 won't power on after the bmc has been relfashed sometimes and requires a hard power off
* The x86 serial ports are not configured by the bmc, so I don't get console output
* The x86 serial ports are muxed through the bmc, so serial config needs to happen *early* for x86 serial consoles to work (aspeed does it in u-boot)
* Fan speed doesn't seem to work

Keno, 2019-09-05:
Yes, I've been using host serial by sshing into the bmc so far, but we should probably have them hooked up to the physical ports by default or at least mirror to the physical ports.

osresearch, 2019-09-05:
Can you document how you access the host serial from the bmc? I wasn't able to get any output.

Keno, 2019-09-05:
Yeah, can do. @bluecmd and I will figure out the serial setup.

c0d3z3r0, 2019-10-26:
I used this to dump the vendors serial setup from the registers ;-) https://github.com/c0d3z3r0/aspeedtool

You need to dump the flash, extract the squashfs, put a cross-compiled telnet/dropbear and aspeedtool into the fs, repack and reflash it ;)

Keno, 2019-10-26:
Cool :), though I think you can also just puppeteer the aspeed from the host and dump the registers without reflashing since the vendor firmware doesn't disable any of those bridges.

c0d3z3r0, 2019-10-26:
I'm not sure if you can access *all* registers via the bridge

Keno, 2019-10-26:
Well, you have access to RAM and the instruction pointer, so you can do ye old debugger trick for running arbitrary code ;).

osresearch, 2019-10-26:
The serial setup is working as of https://github.com/osresearch/u-bmc/commit/8b7d1ae02a33bbfda7487b6a2162a3317a355f00

I have the x86 ttyS0 on the db9 and the BMC console on the internal 10 pin (with supermicro special pinout)