u-root/u-bmc #192: Build in repo directory fails // GOPATH

c0d3z3r0, 2019-09-27:
When building directly in the source / git directory, the build fails:

~~~
cp linux/build/boot/zImage.boot/arch/arm/boot/zImage boot/zImage.boot
go get ./boot/loader/
go get: no install location for directory /home/c0d3/src/ipmi_firmware_tools.git/u-bmc/boot/loader outside GOPATH
	For more details see: 'go help gopath'
make: *** [Makefile:75: boot/loader/loader] Error 1
~~~

bluecmd, 2019-09-27:
Hi!

u-bmc is a go binary and thus requires to be cloned in the Go path. Try cloning to $HOME/go/src/github.com/u-root/u-bmc. Hopefully that will solve your problem.

c0d3z3r0, 2019-09-28:
> requires to be cloned in the Go path

is this a hard requirement of Go? (Well, I read the man page but didn't really find an answer...)

I simply don't want to be forced to clone the repo were Go wants it... I solved this by adding a symlink:

~~~
git clone https://github.com/u-root/u-bmc
cd u-bmc
mkdir -p src/github.com/u-root/
ln -s $PWD src/github.com/u-root/u-bmc
~~~

bluecmd, 2019-09-28:
I think somebody told me that with Go modules that requirement goes away, but sadly yes it's currently a Go requirement as far as I know

c0d3z3r0, 2019-09-28:
Yay, then I have another reason to dislike go :D

Keep up the good work :-)

closed 2019-09-28 by c0d3z3r0

hugelgupf, 2019-09-28:
That requirement is gone with modules, but we don't support modules with u-root yet.