u-root/u-bmc #195: Use a Linux tar file instead of submodule

osresearch, 2019-09-29:
One of the major pains with developing u-bmc is the Linux submodule -- on a slow network it can take tens of minutes to do a clean checkout due to fetching the entire Linux commit history. Ideally it could be a shallow submodule with `--depth 1`, but as I tweeted about my frustration with submodules back at the end of 2017, shallow submodules on branches are still broken:

```
✓ git checkout --depth 1 --branch $BRANCH $URL
✓ git submodule add --branch $BRANCH  $URL
✓ git submodule add --depth 1 $URL
✘ git submodule add --depth 1 --branch $BRANCH $URL
"cannot update paths and switch to branch at the same time"
```

Fetching a tar file from kernel.org and comparing signatures would be 100x faster, plus it would be possible to share the file between builds.

bluecmd, 2019-09-29:
Sure.

This is what the CI is already doing:

```
# Since linux is huge and we're tracking a branch, submodules
# take a long time and we cannot use depth. Fetch the latest zip
# instead
wget https://github.com/openbmc/linux/archive/dev-5.2.zip
unzip dev-5.2.zip
rm dev-5.2.zip
rm -rf linux/
mv linux-dev-5.2/ linux/
(cd linux/; ../linux-patches/apply.sh)
```

I don't see anything stopping from creating a linux target in the Makefile that does all that - allowing users to use git for Linux if they wish while also allowing a fast .zip download.

osresearch, 2019-09-30:
The other reason to use a tar file is that it ensures that we have a reproducible build.  Pulling the latest submodule (without an explicit tag) means that everyone would have a different source tree.

bluecmd, 2019-09-30:
The current submodule is fixed on a specific commit. If you update it as per the readme it should always end up on the specified commit (as of this writing `linux @ f9e04c3`).

The tar archive (actually a zip) I mentioned above would actually be a step backwards in this case, as it would be a zip from the latest head of the development branch. However, nothing is stopping us from using the per-commit zips like `https://github.com/openbmc/linux/archive/f9e04c3157234671b9d5e27bf9b7025b8b21e0d4.zip` more than that it looks horrible.

osresearch, 2019-09-30:
I updated the patch to fetch the tagged tar file and extract it: https://github.com/u-root/u-bmc/pull/198/commits/872773f40030216f367dfe568bfe3b4ca2307089

closed 2019-10-01 by bluecmd