u-root/u-bmc #196: Speed up u-bmc boot

osresearch, 2019-09-29:
Some ideas for speeding up the u-bmc boot time:

* One kernel. Starting the boot kernel that kexec's the u-bmc kernel adds quite a bit of time.
* Read-only root: eliminating the need to validate the flash memory filesystem would save minutes on initial boot, several seconds on normal boots.
* Uncompressed root: xz is 10x the decompression time of gz, and gz adds a few hundred ms.
* Modules: we might be able to save some time by removing what is in the kernel versus loaded later. (NIC, perhaps others)
* Quite boot: printk at 115200 baud is quite slow

On the Supermicro x11ssh, it would be nice to poke the UART bits as early as possible so that the x86 can produce output before the bmc is booted.

bluecmd, 2019-09-29:
I think in terms of speed gain, the read-only root is probably the best to start with. The UBIFS resize and GPG verification takes 90% of the boot time, and we know there are simply better ways to do things.

osresearch, 2019-10-03:
with most of these changes in place and using the slow spispy, the supermicro x11ssh goes from power on to u-bmc prompt in about 16 seconds.
2 seconds for aspeed DDR init
3 seconds to load the kernel from flash (3.5 MB at single spi @ 12 mhz, gz compressed zImage)
2 seconds of printks
7 seconds to start /init from the flash (7.3 MB at single spi @ 20 MHz, squashfs zstd)
2 seconds of log messages
2 seconds to generate sshd key

shenki, 2019-10-08:
re: speeding up the kernel, there was a presentation at Linux Plumbers this year about async probing. Drivers can opt in by setting PROBE_PREFER_ASYNCHRONOUS. I have not looked into this at all.