u-root/u-bmc #204: Make integration a proper platform

osresearch, 2019-10-03:
Currently the integration kernel is special cased in the Makefile.  Should it be a proper platform instead?

ullbeking, 2019-10-03:
This seems like a highly worthwhile thing to consider.

1. What kinds of benefits do you have in mind?
1. How do you expect an integration kernel to differ from a deployed one?  If this question doesn't make sense I guess it indicates I need to learn more about the way that different platforms are specified.
1.  Do you suspect that there are any holes in my knowledge to understand this problem properly?

The reason for my sudden interest in your problem is that I work professionally as an infrastructure engineer and aspriring SRE are as follows.

**I will cut and paste the following paragraphs in to a personal website or profile, where they really belongs**

I am an infrastructure engineer and aspriring SRE who leans slightly more heavily by nature towards the "Dev" side of "DevOps".  This is a coarse way indicating that I'm actively reacquianting myself with my academic background in signal processing and machine learning (PhD 2008).

I have spent the last two years learning the about basics of firmware engineering, particularly as it concerns coreboot and BMC implementations.  Additionally by self-education I am learning the foundations and basics of geo-distributed computing and formalized reasoning.  Naturally these topics benefit the "Ops", hence bringing things full circle.

osresearch, 2019-10-03:
The "integration" platform is for running under qemu x86-64 as part of the CI tests, so it has quite a few differences from the kernels running on the real ARM hardware.  We could try to use qemu-arm, except that they don't have full emulation of the BMC SOC, so it is easier to test the software side on a more fully supported platform.

This is just a code re-org suggestion for consistency (similar to #203), not a major overhaul.