u-root/u-bmc #208: I was just given two Dell i610 power edge servers; and I want to add support for this library

wade-welles, 2020-06-21:
Is a power-edge of this generation desired? I suspect that power-edges with Idrac6 from, or around, the same generation would all be supported by this, or at least provide a simple starting point to continue to extend the functionality. 

We are tearing down the entire server, pulling out all auxiliary and possibly in-securable components, and focusing on documenting the process of locating all the firmware, and flash memory containing important firmware that is either difficult or never changed . 

We are very interested in using both u-root, and u-bmc. 

Do you have any experience with this, do you have any suggestions on where to start reading (code or documentation ) for getting started. 

One method I had considered was dumping the relevant fash memory and implementing a virtual implementation using QEMU or even simpler tools to speed up development and the upside is also being able to offer these tools to others wishing to take the journey of implementing more platforms. 

Love your work, I hope, we can collaborate more, and talk sometime. 

Take care

bluecmd, 2020-06-29:
Hi! Very cool!

The bar I've set for u-bmc is that it's for any BMC that runs a modern Linux. So if you can get Linux ported to the idrac6 you will have the option of running u-bmc (and possibly also OpenBMC) on it.

Starting with a qemu port sounds like a good first step, happy to bounce ideas along the way if you need it.

Best of luck!

Hurricos, 2022-04-23:
For context: iDRAC6 is based on the WPCM450, which [Jonathan Neuschafer has generously spent time porting to Linux](https://github.com/neuschaefer/wpcm450/).